RobertoBenjami/stm32_graphics_display_drivers #2: ili9486

clear-sky, 2019-11-23:
if possible please support ili9486 in 8 bit mode

RobertoBenjami, 2019-11-28:
Please try the ili9488 driver. According to the datasheet, they are very similar.

clear-sky, 2019-12-28:
thank you sir!
it works!

clear-sky, 2019-12-29:
colors is inverted and can see some vertical and horizontal lines
any help?

RobertoBenjami, 2019-12-30:
Upload the project here to watch it.

clear-sky, 2019-12-31:
hi dear roberto
please check your email

RobertoBenjami, 2020-01-02:
I checked, it's interesting. Does the display include a touch screen? If so, how many ohms can be measured between LCD_WR and LCD_D7? What kind of microcontroller and what settings is it running? What length of wire is connected? I checked the datasheet of ili9486, it should work fine.

clear-sky, 2020-01-04:
1.Does the display include a touch screen?
NO
2.If so, how many ohms can be measured between LCD_WR and LCD_D7?
NOP
3.What kind of microcontroller?
STM32F103RE
4.what settings is it running?
my lcd has part namber Z350IT002
datasheet
http://aitendo3.sakura.ne.jp/aitendo_data/product_img/lcd/tft2/Z350IT002/Z350IT002_aitendo.pdf
i have set r8 instead of r16 base on end of page 7 on this datasheet(hardware set) using google translate

pin setting:
CS C10
RS C11
WR C09
RD C08
RST C12
D0---D07 C00---C07
D08---D15 > GND

5.What length of wire is connected?
LESS THAN 20CM

i tryed another driver based on stm32duino
https://github.com/prenticedavid/MCUFRIEND_kbv
i sent picture quality please check your email

and found another bug
in ili9488.h file
#define  ILI9488_ORIENTATION       1
 there is no difference between 1 and 3
also  there is no difference between 2 and 4

thanks for the time you spent for a beginner

closed 2020-01-06 by clear-sky