RobertoBenjami/stm32_graphics_display_drivers #4: Touch support for ili9341?

yuanyanhui, 2020-02-12:
It seems there is no touch support for  ili9341. I tried to use appPaint with ili9341 and found there are no ts_drv and ts_cindex definitions.

RobertoBenjami, 2020-02-14:
I Updated the driver, but I can't tested (I have not a 8 bits paralell ili9341 display with touch). Please first step is the appTouchCalib.

yuanyanhui, 2020-02-26:
My display is different from Adafruit's which has X-/+, Y-/+ pins exposed. The touch controller of my display use SPI connection. Does this mean I cannot use the touch support in your library?

RobertoBenjami, 2020-02-27:
I put the touchscreen on the paralell 8bit, because they are in common pins with the display control and therefore cannot be separated. 
What kind of display is it?

yuanyanhui, 2020-02-29:
[image]
[image]

The touch controller IC is XPT2046. 
[image]

RobertoBenjami, 2020-03-01:
Since this touchscreen is independent of the display controls, any xpt2046 driver can be used. But if I have time, I'll check it out because I have this touch chip on one of my displays.

Rajssss, 2020-12-10:
@yuanyanhui I have a similar module, I am trying to make it work but the display shows nothing, no backlight too. When I pull the LEDA pin to 3.3V the backlight did work but still nothing on the display.

RobertoBenjami, 2022-02-15:
Since then, the standalone xpt2046 driver has been completed.

closed 2023-04-19 by RobertoBenjami