RobertoBenjami/stm32_graphics_display_drivers #5: How to run it on Keil?

JuMalIO, 2020-03-28:
Any one can run this on Keil? I had no luck so far, just black screen...

RobertoBenjami, 2020-03-31:
I modified the driver for keil but couldn't try it because I don't have a licensed version (the 32kB code size limit is too low). Please test it.
The change:
random() -> rand()
_impure_ptr -> optional, I used it just to generate the same pseudorandom numbers on each run
#pragma GCC push_options  -> #pragma push
#pragma GCC optimize("O0") -> #pragma O0
#pragma GCC pop_options -> #pragma pop
beer_60x100_16.c ->I did not find the right solution for this but as a forced solution bmp.h : line39 -> uint16_t data[60 * 100];

JuMalIO, 2020-04-01:
Thanks for fast response!

Project now compiles with Keil, but something is still wrong. After I upload and start the device I see screen being cleared to black color and it stops there, nothing happens after that. If I compile exact same project code with TrueStudio - application runs without issues. Where could be the issue?

Maybe these warnings which I still get are important?
..\Src\Lcd\lcd_io_spi.c(1095): warning:  #61-D: integer operation result is out of range
`GPIOX_MODE(MODE_PP_ALTER_50MHZ, LCD_MOSI);`
..\Src\Lcd\lcd_io_spi.c(1095): warning:  #61-D: integer operation result is out of range
`GPIOX_MODE(MODE_PP_ALTER_50MHZ, LCD_MOSI);`
..\Src\Lcd\lcd_io_spi.c(1095): warning:  #68-D: integer conversion resulted in a change of sign
`GPIOX_MODE(MODE_PP_ALTER_50MHZ, LCD_MOSI);`

RobertoBenjami, 2020-04-04:
printf:
1. delete the syscalls.c file.
2. http://www.keil.com/support/man/docs/uv4/uv4_db_dbg_printf_viewer.htm
Unfortunately, I can’t test because I only have (32k code limit) Keil running in a virtual box.

JuMalIO, 2020-04-06:
Yes, printf is working after setup. Thanks for support. Your drivers are great!

closed 2020-04-06 by JuMalIO

xiaoxinpro, 2024-10-08:
### On the question "error: #1077: an initializer cannot be specified for a flexible array member"

Add '--gnu' in Keil configuration 'C/C++' - 'Misc Controls'

[image]