RobertoBenjami/stm32_graphics_display_drivers #6: Whats the fastest way to draw?

JuMalIO, 2020-04-19:
Hi, I wanted to ask whats the fastest way to draw?
I noticed you are formatting whole BMP image and then sending to draw it.
Whats the best size to send such images? (having in mind that ram cant hold whole image)
Would you be interested in contributions for making font to support dynamic width per char and other changes in stm32_adafruit_lcd?

JuMalIO, 2020-04-19:
Also I have tried to use `BSP_LCD_DrawRGB16Image`. it seems that it does not block while drawing?
Because if I load another image into same RAM from where I draw first image first image gets distorted, because it did not finish drawing and starts drawing the second image in first image place. How could I solve this? Its a bug?

RobertoBenjami, 2020-04-20:
You can set the wait for the end of the drawing in the lcd_spi_io.h file in the #define LCD_DMA_TXWAIT macro. If it is 1 it waits for the drawing to end, if 0 it does not wait. Setting 0 can make a faster program, but be careful not to overwrite the memory that contains the bitmap (or use a double buffer to draw). If extern uint32_t LCD_IO_DmaTransferStatus == 0 then drawing is complete.

RobertoBenjami, 2020-04-20:
The drawing of the letters was not written by me, the bitmap method was inherited from the original adafruit function. I just added some new drawing features that were missing from the original. You can also create a variable-width (proportional) font by modifying stm32_adafruit_lcd. However, aligning the text horizontally will not be easy.

JuMalIO, 2020-04-21:
LCD_DMA_TXWAIT - will this wait block everything even interupts? or it will allow them and will continue as soon as DMA finishes? and wont wait longer (if interupt happened inside its time)?

Also for ili9341 drivers I suggest to add clear display command before turning on display, otherwise it blinks.

RobertoBenjami, 2020-04-23:
It does not block interrupts (why should you block them?), It just waits for the DMA transaction to end (if configured). I have added the configuration options, it is already possible to choose whether to wait for the end of the dma transaction for bitmap and fill operations.
During initialization, clearing the screen is a good idea, I put it in an optional way.

jmalStorm, 2020-04-24:
Yes, if you draw a screen during initialization picture appears instantly when you turn on display, I wonder could similar thing used after you turn on display?

JuMalIO, 2020-04-24:
Tried it kinda work but during display off display does not maintain current display pixels it becomes white... maybe there are another commands that I could try? "ILI9341"

RobertoBenjami, 2020-04-24:
The default state of the display is white. It is also white before initialization. This can only be changed by switching off the backlight. I have 2 types of ili9341 displays. I can't turn off the backlight on the stm32f429 discovery board because it can't be controlled. My other display is a standalone 16-bit parallel, the backlight is on this separate pin, I can control this with a GPIO pin.
Even in sleep or display off mode, the display is white if it is not possible to turn off the backlight.

closed 2020-04-27 by JuMalIO