RobertoBenjami/stm32_graphics_display_drivers #7: using LCD_IO_WriteCmd8MultipleData16 for each image row

JuMalIO, 2020-05-02:
Would it be possible to expose these functions:

    ili9341_SetDisplayWindow(Xpos, Ypos, Xsize, Ysize);
    LCD_IO_WriteCmd8MultipleData16(ILI9341_RAMWR, (uint16_t *)pData, Xsize * Ysize);

So it would be possible to send each row of image without calling ili9341_SetDisplayWindow?

    ili9341_SetDisplayWindow(Xpos, Ypos, Xsize, Ysize);
    for (int i = 0; i < image_height; i++)
    {
      // read image line
      LCD_IO_WriteCmd8MultipleData16(ILI9341_RAMWR, (uint16_t *)pData, Xsize * Ysize);
    }

Just interested to see would it improve performance. I am trying to make screen as fast as possible...

RobertoBenjami, 2020-05-03:
Try this:
`#define ILI9341_RAMWRCONT       0x3C
ili9341_SetDisplayWindow(Xpos, Ypos, Xsize, Ysize);
for (int i = 0; i < image_height; i++)
{
  // read image line
  if(!i)
    LCD_IO_WriteCmd8MultipleData16(ILI9341_RAMWR, (uint16_t *)pData, Xsize * Ysize);
  else
    LCD_IO_WriteCmd8MultipleData16(ILI9341_RAMWRCONT, (uint16_t *)pData, Xsize * Ysize);
}`

JuMalIO, 2020-05-06:
Its amazing thank you again!

closed 2020-05-06 by JuMalIO