RobertoBenjami/stm32_graphics_display_drivers #8: STM32F091RC Nuclio - can it be that DMA interfare with CAN BUS?

JuMalIO, 2020-05-08:
Hi, your driver is working fine with STM32F091RC, but once I try to draw while also sending receiving CAN BUS messages MCU gets into infinite loop in WaitForDmaEnd, WaitForDmaEnd value seems to be 1.

Is my configuration wrong?

My configuration:
     #define LCD_SPI           2
     #define LCD_SPI_MODE      1
     #define LCD_SPI_SPD_WRITE 1
     #define LCD_SPI_SPD_READ  3
     #define LCD_SPI_AFR       0
     #define LCD_RST           B, 11  /* If not used leave it that way */
     #define LCD_RS            B, 14
     #define LCD_CS            B, 12
     #define LCD_SCK           B, 13
     #define LCD_MOSI          B, 15
     #define LCD_MISO          X, 0  /* If not used leave it that way */
     #define LCD_BL            X, 0  /* If not used leave it that way */
     #define LCD_BLON          0
     #define LCD_SCK_EXTRACLK  0
     #define LCD_DMA_TX        1, 5, 3, 0
     #define LCD_DMA_RX        1, 4, 3, 0
     #define LCD_DMA_TXWAIT      0
     #define LCD_DMA_RX_BUFSIZE  256
     #define LCD_DMA_RX_BUFMODE  1

RobertoBenjami, 2020-05-09:
What type of display?
Does it only work incorrectly if the can interface is also running?
Does it work well without DMA?
If no read the bitmap data from the display, then:
```C
#define LCD_DMA_RX 0, 0, 0, 0
```
You can even try channel 7:
```C
#define LCD_DMA_TX 1, 7, 3, 0
```
I only have one f051c8t from the f0 family, so I can't do a practice test with f091.

JuMalIO, 2020-05-09:
ILI9341 display.

In software mode (without DMA) it works fine.

In hardware mode (with DMA) it works only until I get CAN message and on that CAN interrupt I start drawing. It draws first symbol on screen and then waits for DMA to finish and gets stuck in infinite loop.

Tried:
`#define LCD_DMA_RX 0, 0, 0, 0`
But does not make a difference, same loop exists.

Channel 7 seems not to work. Or I need to change anything else, like PINs?
`#define LCD_DMA_TX 1, 7, 3, 0`
Display does not show anything.

JuMalIO, 2020-05-09:
It works on channel 7 only if I define RX channel also (I do not use reading from screen though):

    #define LCD_DMA_TX        1, 7, 3, 0
    #define LCD_DMA_RX        1, 6, 3, 0

but infinent loop still persists on CAN message.

RobertoBenjami, 2020-05-09:
It is not a good idea to run a drawing function from an interrupt. If the DMA interrupt priority of the drawing function is less than that from which the drawing was operated, it will go into an infinite loop. Either the DMA interrupt priority should be set higher than its CAN (#define DMA_IRQ_PRIORITY 15) or do not draw in the interrupt.

JuMalIO, 2020-05-16:
Thanks, I have also tried ILI9486 drivers. It works nicely. Awesome work!

closed 2020-05-16 by JuMalIO