RobertoBenjami/stm32_graphics_display_drivers #9: animation question

pauledd, 2020-05-10:
Hi
Sorry, programming noob alert!
Not a bug, rather a programming question. 
I connected the display to my blue pill via 8bit parallel and works so far.
How would one realize for example a smooth moving rectangle(w=40,h=40) from X0/Y0 to X0/Y200 ?
I tried with a simple for loop counting up to the screens height-rectangleSize, moving the rect step by step and after each step I cleared the screen with BSP_LCD_Clear(RGB888TORGB565(0,0,0)).
But clearing the screen like that is very time intensive and gives obviously not a smooth movement.

My next try was to move the rect and behind I move another rect that clears the previous rect with the background color. That gives me a nice smooth animation instead of clearing the whole screen every step.
Is that the right way? So how would you do that?

RobertoBenjami, 2020-05-11:
Hi.
Without DMA and for framebuffer with little RAM, it is difficult to make continuous motion. For a single square, perhaps the best method is to first draw the new square and then delete the parts of the previous square that are outside the new square. That way, there will be no flicker, and perhaps this can be solved with the least amount of drawing operation. If the square is always the same color, you can even speed it up by drawing only the part of the new square that is outside the previous square.
You might even check out App / 3d_filled_vector. It's pretty tricky. It maps a few lines of frame buffers in memory and exposes them to the LCD. On SPI LCD, 3D animation is quite continuous with the help of DMA, because while DMA is drawing, the processor can generate an image in another frame buffer. This program is faster on the serial SPI interface with DMA than on the parallel 8-bit interface without DMA.

pauledd, 2020-05-12:
> This program is faster on the serial SPI interface with DMA than on the parallel 8-bit interface without DMA.

Oh wow, I didn't know that. I will look into that, thanks a lot for your explanation.

closed 2020-05-12 by pauledd