RobertoBenjami/stm32_graphics_display_drivers #11: Horizontal display with good viewing angles

JuMalIO, 2020-05-17:
Hi, just wanted to ask if you know any horizontal display with good viewing angles?

I have bought this ILI9486 [http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO](url), but I think this display is not intended to be used in horizontal orientation, I was really disappointed, its horizontal viewing angles are horrible.

RobertoBenjami, 2020-05-22:
Perhaps IPS type displays have a wider viewing angle. You may also want to check out the displays for RPI.

JuMalIO, 2020-05-24:
And what about SSD1963 (480x272), do you have any plans to add that? Or ILI9806G (480x854)? Or they are too different?

jmalStorm, 2020-05-25:
I have also found DWIN DGUS Smart Display display, which are IPS, high resolution and quite cheap. They seem to be very fast, but I don't really like that I wont be able to do low level coding on it... Would love to see them hacked for low level application...

RobertoBenjami, 2020-05-27:
I don't know these displays. Larger displays are already expensive. If I need one, I'd rather use an old 15 'or 17' monitor with an analog VGA connector. They are easy to operate with LTDC capable controllers.

closed 2020-05-29 by JuMalIO