RobertoBenjami/stm32_graphics_display_drivers #13: Image distortions on STMF407VE

JuMalIO, 2020-05-21:
Hi when I moved same code from STM32F091RC to STM32F407VE (changed driver to FMCS 8bit) I am getting strange image distortions. On STM32F091RC everything was working fine.
I am using ILI9486 driver. Tried no DMA but does not change images apear distorted.
Any ideas?

JuMalIO, 2020-05-22:
GPIO8 solution seems to be working ok, the issue is with FMCS 8bit on Keil, for some reason lines starts to shift and images get distorted, It draws too fast or something? Or its STM32F407VE memory issues? How could I check whats worong?

RobertoBenjami, 2020-05-22:
In CUBE, the FSMC configuration, you should experiment with these values:
- Address setup time in HCLK clock cycles
- Data setup time in HCLK clock cycles
- Bus turn around time in HCLK clock cycles
Good values also depend on the cpu frequency, display, and wire length.

jmalStorm, 2020-05-25:
in deed I am using ~50cm wires to display. FSMC seems to be very sensitive....

RobertoBenjami, 2020-05-27:
The parallel system bus is not designed to bridge 60cm distances with it. It may be possible to operate flawlessly at this distance, only with the right cable and proper impedance matching.

closed 2020-05-29 by JuMalIO