RobertoBenjami/stm32_graphics_display_drivers #15: stm32f407ve LL spi possible?

pauledd, 2020-06-20:
Hi

Is it possible to drive my ILI9341 (DSO138) display on the STM32f407VET board (black)
via LL drivers without HAL using spi dma?

Or does this require heavy rewriting of your driver code to work with LL?

RobertoBenjami, 2020-06-22:
The DSO138 does not have an SPI but a parallel 8-bit ILI9341 display. If you have an SPI ILI9341 display, there is also an example video for stm32f429. It must be set in the same way on stm32f407. HAL is used only during initialization, there only the HAL_Delay function. Only this needs to be replaced with another delay function.

pauledd, 2020-06-22:
Well, I somehow missed that there is no SPI on that display... thanks for correcting me. I guess I'll get an SPI display because I dont need high update rate and I dont want that much cables used in 8bit mode

closed 2020-06-22 by pauledd