RobertoBenjami/stm32_graphics_display_drivers #19: Add font?

reyfrancis, 2020-10-03:
Hi. Thanks for your hard work on this library. I want to ask on how can we add a custom font and font size.

RobertoBenjami, 2020-10-21:
1. copy one of the C files to the font folder, give it your own name (eg copy font24.c myfont24.c)
2. open the myfont24.c file
3. modify: const uint8_t Font24_Table -> const uint8_t MyFont24_Table
4. modify: sFONT Font24 = {Font24_Table -> sFONT MyFont24 = {MyFont24_Table,
5. modify the character bitmap patterns
6. Open the fonts.h file
7. add the following: extern sFONT MyFont24;
8. use it
You can also change the font size in the c file:
sFONT MyFont24 = {
   Font24_Table,
   17, / * Width * /
   24, / * Height * /
};

TDMLab, 2021-06-25:
Hi Roberto. Thank you so much for writing the library. I checked adding new fonts and everything works for a 16x32 pixel font, but when I wanted to insert a larger 32x48 font, it was no longer displayed. 
I did everything as described above and I also found this piece of code in stm32_adafruit_lcd.c, 
`#define MAX_HEIGHT_FONT       48
#define MAX_WIDTH_FONT        32
#define OFFSET_BITMAP         54

/* Max size of bitmap will based on a font24 (17x24) */
static uint8_t bitmap[MAX_HEIGHT_FONT * MAX_WIDTH_FONT * 2 + OFFSET_BITMAP] = {0};`
I didn't quite understand what it does and set the maximum font sizes, but it still didn't work.

RobertoBenjami, 2022-02-15:
Can you put an example project somewhere with such a large character set?

RobertoBenjami, 2022-02-15:
Is there a character set file that didn't work well? Can you share it?

closed 2023-04-19 by RobertoBenjami