RobertoBenjami/stm32_graphics_display_drivers #24: Driver integration with STemWin library

iTano86, 2021-04-01:
Hello Roberto. First of all I want to thank you for sharing your drivers. They are really very useful.
At the moment I'm using the driver for ILI9486 with an F446RE nucleo board (8 bit parallel with Touch screen) and it works perfectly.
However, I cannot integrate your drivers with the STemWin library.
Could you help me?
Thanks again

RobertoBenjami, 2021-04-30:
The stemwin library is not open source. It only has headers and a library. This makes it very difficult to fit.

iTano86, 2021-05-03:
Roberto thank you for your reply. 
I apologize for having replied late (unfortunately I had some problems that did not allow me to stay at the PC in the previous days); I integrated, in some way, the Stemwin library but the functioning is not optimal. 
I will try to create a simple GUI using the functions built into your firmware. 
Thanks again

closed 2022-02-15 by RobertoBenjami