RobertoBenjami/stm32_graphics_display_drivers #25: License

Petr019, 2021-04-21:
Hello Roberto,

thanks for the drivers, they are great!

I would like to use one of the drivers in closed source project and I did not found any license disclaimer in your repository. 
As discussed e.g. here 
https://opensource.stackexchange.com/questions/1720/what-can-i-assume-if-a-publicly-published-project-has-no-license
no license means that all rights are reserved and I cannot modify or redistribute this code without explicit permission from the copyright holder. 

Could you please state the license type or could I get permission to use it?

Thanks,
Petr

AmirMahdiHassanzadeh, 2021-04-22:
Hi, I am Amir Mehdi, my code is free and you can make any changes you like to these sources, and of course I am happy that the open source community is growing.  I will definitely put a license based on this soon

RobertoBenjami, 2021-04-30:
Use it. I allow it.

Petr019, 2021-05-03:
> Komu: RobertoBenjami/stm32_graphics_display_drivers <stm32_graphics_display_
> "

OK. Thanks!

Petr 

---------- Původní e-mail ----------

(https://github.com/RobertoBenjami/stm32_graphics_display_drivers/issues/25#issuecomment-829923893)
"

closed 2022-02-15 by RobertoBenjami