RobertoBenjami/stm32_graphics_display_drivers #26: 'GPIOX' undeclared (first use in this function); did you mean 'GPIOA'?

Hamed-Elahi, 2021-05-23:
Thank you for the code. I've done everything like the video, but there is some errors in this file "lcd_io_gpio16.c".

For example, in line 73 (#define GPIOX_SET_(a, b)      GPIO ## a ->BSRR = 1 << b), there is this error:

'GPIOX' undeclared (first use in this function); did you mean 'GPIOA'?

I've commented these 5 lines to see other possible errors. Now, there is an error in this file "lcd_io_gpio16.h".

In line 8 (#define LCD_CS            F, 13), there is this error:

'F' undeclared (first use in this function)

Do you know what I've done wrong?

Thank you in advance.

Hamed

RobertoBenjami, 2021-05-23:
Which processor family? Can you share this code?

Hamed-Elahi, 2021-05-24:
Hi, I'm using NUCLEO-L552ZE-Q with STM32L552ZE MCU.

Yes, I attached the code.

[LCD_HY32D_with_touchscreen.zip](https://github.com/RobertoBenjami/stm32_graphics_display_drivers/files/6529773/LCD_HY32D_with_touchscreen.zip)

RobertoBenjami, 2021-05-24:
1. cannot be compiled until pins assignments are undefined in the lcd_io_gpio16.h file (modify the X, 0)
2. the lcd_io_gpio16 driver cannot touchscreen
3. the ts_xpt2046 driver is required (currently only versions F2 and F4 family)

Hamed-Elahi, 2021-05-24:
I modified one time all the pins, but it didn't work. So, there might be some compatibility issue or two other cases that you mentioned. 

Thank you so much, with these provided information, maybe I can find some solutions or other drivers which work with this LCD.

The main problem is: this LCD "HY32D" is an old TFT LCD which I couldn't find some stm32 driver for it.

RobertoBenjami, 2021-05-24:
You may want to try it first without a touchscreen (e.g. with the LcdSpeedTest App). If that already works, you can deal with the touchscreen. You may also want to try the ili9341 driver.
I only have ili9341 parallel 16 bit display.

Hamed-Elahi, 2021-05-24:
Thank you, I will do it. When there is something new, I will inform you.

Hamed-Elahi, 2021-05-24:
Wow. I cannot believe it. It works perfectly. Now, maybe there is a problem with clock or something else. I must change and try them.

You were right with the error. Maybe, the first time that I assigned the pins, I haven't assigned LCD_BL, because of this comment "/* If not used leave it that way */". That was my fault. Thank you so much again.

As you told, I could run it now with "appLcdSpeedTest". And I used "ili9325.c" and "lcd_io_gpio16.c" to run it. So, now I know that ili9325 driver does work without problem, but I must also try more times with "appTouchCalib". When I've tried it, there is an error. At first, I try alone to find the problem, but if I cannot solve it, let you know.

Have a nice week!

Hamed

RobertoBenjami, 2021-05-24:
I created the ts_xpt2046 touchscreen driver for L5. I can't try it because I don't have an L5 chip. Try. Applications that use the touchscreen also need a touchscreen driver. Remember to configure the pins in ts_xpt2046.h.
LCD_BL, LCD_RD, LCD_RST can remain X, 0 if not used (written there). The other pins must be adjusted.

Hamed-Elahi, 2021-05-24:
Ok, I will do it. Thanks.

Hamed-Elahi, 2021-06-02:
Hello,

I've tested all you said and many other parts of the code. There is firstly one problem, when I can solve it, then I can test it with the calibration code again.

I attach a video. Please, see this video (there is also a picture, if the video doesn't work). I used this function "BSP_LCD_DrawLine(x1, y1, x2, y2);" with these inputs:

    x1 = 50;
    y1 = 55;
    x2 = 220;
    y2 = 200;

So, the line must be an inclined line, but this line is straight. Because of that, when your code begins and wants to test the lines etc., all the lines are straight and non of them has an inclination.

I've changed many lines of code, used other functions from other codes and changed the registers, but non of them worked. I thought, maybe the problem is on this line:

#define ILI9325_SETCURSOR(x, y)              {LCD_IO_WriteCmd16(RC(ILI9325_GRAM_HOR_AD)); LCD_IO_WriteData16(RD(x)); LCD_IO_WriteCmd16(RC(ILI9325_GRAM_VER_AD)); LCD_IO_WriteData16(RD(y));}

Or there is not enough delay for writing on registers or so on. But I couldn't do anything so far.

What do you think? How can I solve this problem?

To see the line, look at bottom of the LCD in image or video.

[image]

https://user-images.githubusercontent.com/69258588/120518069-739a3800-c3d1-11eb-8544-af613338b1d6.mp4

closed 2022-02-15 by RobertoBenjami