RobertoBenjami/stm32_graphics_display_drivers #28: ILI9341 3.2inch TFT with XPT2046 Problem

Ex1a, 2021-10-02:
Hi Roberto!
I use STM32F103C8T6 and ILI9341 display(SPI) with XPT2046, I tested that all functions can work except Display char function, I cannot display any char at the display, and the Char Test in appLcdSpeedTest is also doesn't work, it just display the black screen.

I tried to change the SPI Baud Rate but it just can control the LCD speed.

Can you tell me how to solve this problem?

Thank you!!

RobertoBenjami, 2021-10-04:
Is there a font(s) in the project?

Ex1a, 2021-10-04:
Yes, I checked it has font.c and font.h.

Here is my project link:
https://drive.google.com/drive/folders/1hCQ0F3AZuFr-VJFC6rHmfvMMyHOMDWBu?usp=sharing

Thank you!!

RobertoBenjami, 2021-10-04:
If you delete lines 530..534 (ili9341_DravBitmap) in the ili9341.c file, you will not be able to write characters to the screen (and a HardFault error will occur because the uninitialized pointer points to a non-existent address).

Ex1a, 2021-10-05:
Oh no, I forgot to tell you, sorry...
[image]

If I run this code, it will give an error.
Thank you!!

RobertoBenjami, 2021-10-05:
"bmp.h" file is missing from the LCD folder. Put it in.

Ex1a, 2021-10-06:
Thank you very much!!