RobertoBenjami/stm32_graphics_display_drivers #30: Help for XPT2046 on ILI9486(SPI mode)

hakkican, 2022-02-05:
Hi.
ILI9488 (spi mode-DMA) works great on STM32F407VET6. I couldn't find any example related to XPT2046. I decided to do it myself. Many compilation errors occurred when I included the XPT2046 touchscreen. The problem appears in the file "lcdts_io_spi_xpt2046.c".
Where am I doing wrong?

Errors:

15:12:54 **** Incremental Build of configuration Debug for project Stm32F407_ili9488_Touch_2 ****
make -j6 all 
arm-none-eabi-gcc "../Core/Src/Lcd/lcdts_io_spi_xpt2046.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F407xx -c -I../Core/Inc -I../Src/Lcd -I../Core/Src/Lcd -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/Lcd/lcdts_io_spi_xpt2046.d" -MT"Core/Src/Lcd/lcdts_io_spi_xpt2046.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Core/Src/Lcd/lcdts_io_spi_xpt2046.o"
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:422:28: error: macro "GPIOX_PORTNUM_" requires 2 arguments, but only 1 given
  422 | #if GPIOX_PORTNUM(LCDTS_SCK) < GPIOX_PORTNUM_A
      |                            ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:129: note: macro "GPIOX_PORTNUM_" defined here
  129 | #define GPIOX_PORTNUM_(a, b)  GPIOX_PORTNUM_ ## a
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:423:2: error: #error not definied the LCDTS SCK pin
  423 | #error  not definied the LCDTS SCK pin
      |  ^~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:426:29: error: macro "GPIOX_PORTNUM_" requires 2 arguments, but only 1 given
  426 | #if GPIOX_PORTNUM(LCDTS_MOSI) < GPIOX_PORTNUM_A
      |                             ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:129: note: macro "GPIOX_PORTNUM_" defined here
  129 | #define GPIOX_PORTNUM_(a, b)  GPIOX_PORTNUM_ ## a
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:427:2: error: #error not definied the LCDTS MOSI pin
  427 | #error  not definied the LCDTS MOSI pin
      |  ^~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:430:29: error: macro "GPIOX_PORTNUM_" requires 2 arguments, but only 1 given
  430 | #if GPIOX_PORTNUM(LCDTS_MISO) < GPIOX_PORTNUM_A && LCD_SPI_MODE == 2
      |                             ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:129: note: macro "GPIOX_PORTNUM_" defined here
  129 | #define GPIOX_PORTNUM_(a, b)  GPIOX_PORTNUM_ ## a
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:431:2: error: #error not definied the LCDTS MISO pin
  431 | #error  not definied the LCDTS MISO pin
      |  ^~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c: In function 'LcdDirRead':
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:674:34: error: macro "GPIOX_MODER_" requires 3 arguments, but only 2 given
  674 |   GPIOX_MODER(MODE_OUT, LCDTS_SCK);
      |                                  ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:94: note: macro "GPIOX_MODER_" defined here
   94 | #define GPIOX_MODER_(a,b,c)   GPIO ## b->MODER = (GPIO ## b->MODER & ~(3 << (2 * c))) | (a << (2 * c));
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:95:31: error: 'GPIOX_MODER_' undeclared (first use in this function)
   95 | #define GPIOX_MODER(a, b)     GPIOX_MODER_(a, b)
      |                               ^~~~~~~~~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:674:3: note: in expansion of macro 'GPIOX_MODER'
  674 |   GPIOX_MODER(MODE_OUT, LCDTS_SCK);
      |   ^~~~~~~~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:95:31: note: each undeclared identifier is reported only once for each function it appears in
   95 | #define GPIOX_MODER(a, b)     GPIOX_MODER_(a, b)
      |                               ^~~~~~~~~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:674:3: note: in expansion of macro 'GPIOX_MODER'
  674 |   GPIOX_MODER(MODE_OUT, LCDTS_SCK);
      |   ^~~~~~~~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:677:24: error: macro "GPIOX_ODR_" requires 2 arguments, but only 1 given
  677 |     GPIOX_ODR(LCDTS_SCK) = 0;
      |                        ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:106: note: macro "GPIOX_ODR_" defined here
  106 | #define GPIOX_ODR_(a, b)      BITBAND_ACCESS(GPIO ## a ->ODR, b)
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:107:31: error: 'GPIOX_ODR_' undeclared (first use in this function)
  107 | #define GPIOX_ODR(a)          GPIOX_ODR_(a)
      |                               ^~~~~~~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:677:5: note: in expansion of macro 'GPIOX_ODR'
  677 |     GPIOX_ODR(LCDTS_SCK) = 0;
      |     ^~~~~~~~~
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:679:24: error: macro "GPIOX_ODR_" requires 2 arguments, but only 1 given
  679 |     GPIOX_ODR(LCDTS_SCK) = 1;
      |                        ^
[... 91 lines cut ...]
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:91: note: macro "GPIOX_AFR_" defined here
   91 | #define GPIOX_AFR_(a,b,c)     GPIO ## b->AFR[c >> 3] = (GPIO ## b->AFR[c >> 3] & ~(0x0F << (4 * (c & 7)))) | (a << (4 * (c & 7)));
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:1402:37: error: macro "GPIOX_MODER_" requires 3 arguments, but only 2 given
 1402 |   GPIOX_MODER(MODE_ALTER, LCDTS_MOSI);
      |                                     ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:94: note: macro "GPIOX_MODER_" defined here
   94 | #define GPIOX_MODER_(a,b,c)   GPIO ## b->MODER = (GPIO ## b->MODER & ~(3 << (2 * c))) | (a << (2 * c));
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:1403:38: error: macro "GPIOX_AFR_" requires 3 arguments, but only 2 given
 1403 |   GPIOX_AFR(LCDTS_SPI_AFR, LCDTS_MISO);
      |                                      ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:91: note: macro "GPIOX_AFR_" defined here
   91 | #define GPIOX_AFR_(a,b,c)     GPIO ## b->AFR[c >> 3] = (GPIO ## b->AFR[c >> 3] & ~(0x0F << (4 * (c & 7)))) | (a << (4 * (c & 7)));
      | 
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:1404:37: error: macro "GPIOX_MODER_" requires 3 arguments, but only 2 given
 1404 |   GPIOX_MODER(MODE_ALTER, LCDTS_MISO);
      |                                     ^
../Core/Src/Lcd/lcdts_io_spi_xpt2046.c:94: note: macro "GPIOX_MODER_" defined here
   94 | #define GPIOX_MODER_(a,b,c)   GPIO ## b->MODER = (GPIO ## b->MODER & ~(3 << (2 * c))) | (a << (2 * c));
      | 
make: *** [Core/Src/Lcd/subdir.mk:28: Core/Src/Lcd/lcdts_io_spi_xpt2046.o] Error 1
"make -j6 all" terminated with exit code 2. Build might be incomplete.

15:12:56 Build Failed. 30 errors, 0 warnings. (took 2s.120ms)

RobertoBenjami, 2022-02-06:
Are all the required headers and c sources copied to the project folder?

hakkican, 2022-02-08:
Project folder like this

C:\CAN\CUBEMXTEST\STM32F407_ILI9488_TOUCH_2\CORE\SRC
│   main.c
│   stm32f4xx_hal_msp.c
│   stm32f4xx_hal_timebase_tim.c
│   stm32f4xx_it.c
│   syscalls.c
│   sysmem.c
│   system_stm32f4xx.c
│
├───App
│       appPaint.c
│
└───Lcd
    │   bmp.h
    │   ili9488.c
    │   ili9488.h
    │   lcd.h
    │   lcdts_io_spi_xpt2046.c
    │   lcdts_io_spi_xpt2046.h
    │   stm32_adafruit_lcd.c
    │   stm32_adafruit_lcd.h
    │   stm32_adafruit_ts.c
    │   stm32_adafruit_ts.h
    │   ts.h
    │
    └───Fonts
            font12.c
            font16.c
            font20.c
            font24.c
            font8.c
            fonts.h
            Release_Notes.html

RobertoBenjami, 2022-02-08:
ili9488.h:
#define  ILI9488_TOUCH  1

ili9488.c - clear this 3 lines:
#if ILI9488_INTERFACE == 0
#include "lcd_io_spi.h"
#endif

I will fix that too.

hakkican, 2022-02-08:
Thanks.
No more compilation errors with this edit.

Touch did not work in the first test. I disconnected the MISO of the display, now it works.
The problem now is that the direction of the touch is reversed. I added a video for it. I changed the value of ILI9488_ORIENTATION but that didn't work. What should I do?

https://user-images.githubusercontent.com/9567968/153002184-89b01b35-4f53-4415-8a84-9e834cb45d9a.mp4

RobertoBenjami, 2022-02-09:
Each touch screen is different. Step 1: Calibration required. You can do this with appTouchCalib.c. Here's a video on how to do it. This will allow you to generate the cIndex values for the display, which will need to be inserted into ili9488.h.
[https://www.youtube.com/watch?v=qyCctzAbD2g](url)

hakkican, 2022-02-11:
Problem solved. Thank you very much.