RobertoBenjami/stm32_graphics_display_drivers #37: FSMC changed RS line

freetoair, 2022-10-30:
I had to change the RS line in the FSMC driver to A17 (PD 12). But I can't find how I should change #define LCD_REGSELECT_BIT 6 to get the proper memory bank select address.
My question is how to get LCD_REGSELECT_BIT =" 6 " in a function of LCD RS select line ?

regards, freetoair.

freetoair, 2022-10-30:
And yes, the ILI9488 works great in FSMC 16bit mode, although I had to set the LCD_ADDR_BASE based on the existing firmware for that development board.
 It would be very useful to know how to get the value of the variable LCD_REGSELECT_BIT

RobertoBenjami, 2022-11-14:
Regselect bit = address line number to which the LCD RS pin is connected (e.g. if it is connected to pin A17, then
#define LCD_REGSELECT_BIT 17)

freetoair, 2022-11-24:
OK, thanks, now it works as it should. I don't know if I should open a new post or can I ask here? Namely, in fsmc mode 8bit with the SPFD 5408 controller, I can't get a display on the screen, only a white screen, no image. This controller in the sequence to write the index register does not need to write the higher byte, only the low uint8_t byte, could it be a problem or the speed of the transaction?

regards, freetoair

closed 2022-11-24 by freetoair

RobertoBenjami, 2022-11-28:
8 bit index register:
- LCD_IO_WriteCmd8MultipleData8 : 8 bit index + 8 bit data (for settings)
- LCD_IO_WriteCmd8MultipleData16 : 8 bit index + 16 bit data (for coordinatas setting and draw bitmap)
- LCD_IO_WriteCmd8DataFill16 : 8 bit index + 16 bit fill data (for draw fill)
FSMC speed settings is possible in CubeMX or CubeIDE (clock cycles parameters)